mholgatem/GPIOnext #99: Audio doesn't work after installation.

mikejkelley, 2025-01-30:
I'm running this on a Pi 5 with a RPI HiFi AMP HAT TAS5713 Amplifier Audio Module 25W Class-D Power Sound Card Expansion Board for Raspberry Pi 5 4 3 B+ Pi Zero Nichicon Capacitor and some custom scripts. After install I can get the joystick to work but my audio is nuked; not only is there no sound but basic amixer commands like amixer -c sset 'Master' 60% have become invalid. Alsa commands are not found.

mholgatem, 2025-01-30:
You need to use the 'set pins' command ( see documentation) if you are
going to use something else in conjunction with gpionext. Ymmv

Not sure about your alsa commands, gpionext doesn't do anything with alsa.

mikejkelley, 2025-01-31:
Hi, I used the set pins command this time but it still wrecks all my audio. Before GPIONext installation amixer -c sset 'Master' 60% works, after installation it becomes invalid. After uninstalling GPIONext amixer -c sset 'Master' 60% commands work again. I've been troubleshooting this for days and am super sad now like this -> :(

mholgatem, 2025-01-31:
With your HAT off (or see *), try setting only pin 11 (which is gpio 17). Just configure a single button to it. then connect the HAT and restart your system. Check if your HAT is working and the button works. That will tell us if there is a configuration error or not.

you can also use `sudo raspi-gpio get` to check the status of the pins. I don't know what the software that your HAT uses does, it might be grabbing pins that it doesn't need. 

I unfortunately don't have a lot of time right now to dive into this, otherwise I'd probably grab a HAT myself and check it out. 

*FYI, the 'set pins' command doesn't apply to the configuration manager, you need to use `gpionext config --pins 11` to set it with just pin 11. Also, pins are board numbered, not BCM.

mikejkelley, 2025-01-31:
I can get the joystick to work or I can get audio to work, but not both at the same time...

mholgatem, 2025-01-31:
This would be my guess: based on the info provided about the HAT on amazon, it shouldn't be using any of those pins. So it is either incorrect, and actually is using 1 or all of those pins, and thus has a problem when gpionext is there. Or, it is using a method incompatible with gpionext to grab those pins. 

I would imagine that whatever is causing the problem between the two, crashes the software of the HAT and causes a crash in ALSA as well, which is why the commands don't work when gpionext is running.

Again, I would recommend testing it with just 1 button/1 pin, not a whole joystick, so that you minimize possible overlap between the two softwares/pins. But in the end, they may just not be compatible.

mikejkelley, 2025-02-01:
This time I had music playing and it cut out the moment I `bash GPIOnext/install.sh`'d. Does it grab all the pins until and unless they're specified? I tried setting pins individually but couldn't get any I tired to work with audio so maybe the documentation is correct? This dooms my project as there isn't another way other than the hat to pipe music to my countercade's speakers from the Pi 5. I'll have to downgrade to a Pi 4 and use analogue audio or use some USB jank and shove everything in the cabinet. [https://youtu.be/Y0_MV_AAKhM](url)

mholgatem, 2025-02-01:
If it does it when you run the install script then that wouldn't be a clash
with pins. The install script doesn't do anything but install packages and
set up udev rules for virtual controllers.

You could go line by line and manually do the install until you find the
problem. There's probably a conflict somewhere.

ld21, 2025-11-04:
I have a similar issue. My sound (using I2S and MAX98357A) on a Raspberry Pi 4 with Bookworm works well. As soon as I install GPIOnext as described, even when limiting it to certain pins (37,8,33,36,29,31), the config-tool raises the following error and sound stopps.
```
Traceback (most recent call last):
  File "/home/toddd/GPIOnext/config_manager.py", line 315, in <module>
    ConfigurationManager(args)
  File "/home/toddd/GPIOnext/config_manager.py", line 86, in __init__
    self.getControllerType()
  File "/home/toddd/GPIOnext/config_manager.py", line 140, in getControllerType
    self.configureKeyboard( currentDevice )
  File "/home/toddd/GPIOnext/config_manager.py", line 199, in configureKeyboard
    self.getControllerType()
  File "/home/toddd/GPIOnext/config_manager.py", line 135, in getControllerType
    gpio.cleanup()
  File "/home/toddd/GPIOnext/config/gpio.py", line 93, in cleanup
    GPIO.cleanup(pinList)
  File "/usr/lib/python3/dist-packages/RPi/GPIO/__init__.py", line 619, in cleanup
    lgpio.gpio_claim_input(_chip, gpio, lgpio.SET_PULL_NONE)
  File "/usr/lib/python3/dist-packages/lgpio.py", line 755, in gpio_claim_input
    return _u2i(_lgpio._gpio_claim_input(handle&0xffff, lFlags, gpio))
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/lgpio.py", line 458, in _u2i
    raise error(error_text(v))
lgpio.error: 'GPIO busy'```

I didn't have that issue in the previous version «Bullseye». What could it be?
Regards, Dani