mholgatem/GPIOnext #100: Detecting

NikhilKonduru8, 2025-03-06:
When i go through the configuration and get to the detecting part. It is not detecting the buttons, it sometimes does but other times it is not.

NikhilKonduru8, 2025-03-06:
I dont know if this is relevant, but im using a usb drive instead of a sd card. will this effect anything. Like the speed or something while detecting. Do you think that is the problem?

mholgatem, 2025-03-06:
It depends on a lot of factors. Are you having problems? If so, we need a
lot more information about your setup.

NikhilKonduru8, 2025-03-06:
I am having problems with detecting the buttons in the config setup. My setup is a raspberry pi 4 4gb, running retro pie. I usually use an sd card but for testing I'm just using a usb drive temporarily. I'm also using generic push buttons for the test controller.

mholgatem, 2025-03-06:
generic push buttons are fine, I use them myself. During the config, you do
have to press and hold the buttons for a second or two before it registers.
A usb drive can lead to gpio instability. If you have a weak power supply
and/or power hungry usb peripherals, it can draw enough power that it drops
the voltage on the gpio, leading gpionext to not be able to register button
presses. I would keep that in mind, but it wouldn't necessarily be the
first thing that I suspect. Are you getting anything from the config when
you try to set it up? I would double check your wiring. Maybe test just a 1
or two button setup first and see where that gets you. Also, remember that
getting through the gpionext config does not guarantee that it will show up
in retropie. If I remember correctly, you need at minimum a full
directional (dpad/joystick), and 2 or 3 buttons. Otherwise retropie seems
to ignore it.

NikhilKonduru8, 2025-03-06:
When detecing the buttons in the config it detects one of the buttons most of the time. But the others ones are not being detected. I don't have any other USB peripherals connected other than a HDMI cable connected to my monitor.

mholgatem, 2025-03-06:
Strange, can you take a picture of your wiring?

NikhilKonduru8, 2025-03-06:
I am not home right now so I can't take the picture, but I will be home in a couple of hours. But I do have a question. Do I need to connect a resistor to push button and the raspberry pis ground for it to work, or can I use a direct connection.

NikhilKonduru8, 2025-03-07:
[image]

here is an image of my wiring. Only the right button is being detected right now. Just saying, im using a pi sugar 2 plus for the power.

mholgatem, 2025-03-07:
Ok, your left button is on the serial UART transfer pin, which may be the cause of it. The easiest method to check would be move it to a different pin. GPIO 17 & 27 are both safe pins. If that works, then you may have serial UART enabled (possibly due to your pi sugar 2 plus). 

You can also try this: 
`gpionext stop` 
`sudo cat /sys/kernel/debug/gpio`
look for gpio-14. If you see something like `gpio-14 (|txd                 ) out hi` then the pin may be in use by something else.

If you run `sudo raspi-config`, then go `interface options > serial port` and answer 'NO' to both questions, that will disable the serial UART.

NikhilKonduru8, 2025-03-08:
Only one of them still works for some reason.