mholgatem/GPIOnext #101: The "Can't add edge detection for pin" error again

BaltasarParreira, 2025-03-10:
Running latest Retropie (buster) on a Raspberry Pi Zero 2W with out any of the intend to use GPIO used by any driver as this can show it:
```
pi@retropie:~ $ sudo cat /sys/kernel/debug/gpio
gpiochip0: GPIOs 512-565, parent: platform/3f200000.gpio, pinctrl-bcm2835:
 gpio-512 (ID_SDA              )
 gpio-513 (ID_SCL              )
 gpio-514 (GPIO2               )
 gpio-515 (GPIO3               )
 gpio-516 (GPIO4               |sdmode              ) in  lo 
 gpio-517 (GPIO5               )
 gpio-518 (GPIO6               )
 gpio-519 (GPIO7               )
 gpio-520 (GPIO8               )
 gpio-521 (GPIO9               )
 gpio-522 (GPIO10              )
 gpio-523 (GPIO11              )
 gpio-524 (GPIO12              )
 gpio-525 (GPIO13              )
 gpio-526 (GPIO14              )
 gpio-527 (GPIO15              )
 gpio-528 (GPIO16              )
 gpio-529 (GPIO17              )
 gpio-530 (GPIO18              )
 gpio-531 (GPIO19              )
 gpio-532 (GPIO20              )
 gpio-533 (GPIO21              )
 gpio-534 (GPIO22              )
 gpio-535 (GPIO23              )
 gpio-536 (GPIO24              )
 gpio-537 (GPIO25              )
 gpio-538 (GPIO26              )
 gpio-539 (GPIO27              )
```

but trying to setup some home made joypad with :
`sudo python3 /home/pi/GPIOnext/config_manager.py --pins 3,5,7,8,10,11,13,15,29,31,32,33`
nothing is detected.

Also setting the pins I want with the:
`gpionext set pins 3,5,7,8,10,11,13,15,29,31,32,33`

I get the flowing errors on the "log/journal":
```
Mar 10 05:11:14 retropie python3[1165]: Can't add edge detection for pin 3(pin is already in use). Skipping.
Mar 10 05:11:14 retropie python3[1165]: Can't add edge detection for pin 5(pin is already in use). Skipping.
Mar 10 05:11:14 retropie python3[1165]: Can't add edge detection for pin 7(pin is already in use). Skipping.
Mar 10 05:11:14 retropie python3[1165]: Can't add edge detection for pin 8(pin is already in use). Skipping.
Mar 10 05:11:14 retropie python3[1165]: Can't add edge detection for pin 10(pin is already in use). Skipping.
Mar 10 05:11:14 retropie python3[1165]: Can't add edge detection for pin 11(pin is already in use). Skipping.
Mar 10 05:11:14 retropie python3[1165]: Can't add edge detection for pin 13(pin is already in use). Skipping.
Mar 10 05:11:14 retropie python3[1165]: Can't add edge detection for pin 15(pin is already in use). Skipping.
Mar 10 05:11:14 retropie python3[1165]: Pin 29 is not set as input. Skipping.
Mar 10 05:11:14 retropie python3[1165]: Pin 31 is not set as input. Skipping.
Mar 10 05:11:14 retropie python3[1165]: Pin 32 is not set as input. Skipping.
Mar 10 05:11:14 retropie python3[1165]: Pin 33 is not set as input. Skipping.
```

All is updated with:
```
sudo apt update
sudo apt upgrade

Other tests I did are working, like doing:
`sudo raspi-gpio get <GPIO Number>`
```
with and without pressing the joypad buttons and the "level" changes ok from 1 (not pressed) to 0 (pressed).

So what can be done?
Is there a new bug with the latest Packages from Buster SO, Retropie or GPIOnext software?

mholgatem, 2025-03-10:
As far as I'm aware, there are no new conflicts. I'm out of town right now.
But I can look into it later in the week.

BaltasarParreira, 2025-03-10:
New update and almost there...
I had kernel 6.12 since I used the rpi-update command, and looks like this is not working on that new kernel so I though and went back to the stable and normal one, the 5.10, and "voila" now the config tool already detects some gpio pins where I have the buttons, but strangely not all since I still get errors on log/journal:
```
Mar 10 22:44:42 retropie python3[1273]: Pin 29 is not set as input. Skipping.
Mar 10 22:44:42 retropie python3[1273]: Pin 31 is not set as input. Skipping.
Mar 10 22:44:42 retropie python3[1273]: Pin 32 is not set as input. Skipping.
Mar 10 22:44:42 retropie python3[1273]: Pin 33 is not set as input. Skipping.
```

I have the config for all the fixed pins I use only with this:
`gpionext set pins 3,5,7,8,10,11,13,15,29,31,32,33`

and I'm doing the setup config with this:
`sudo python3 /home/pi/GPIOnext/config_manager.py --pins 3,5,7,8,10,11,13,15,29,31,32,33`

Also I know that all is ok and I'm not using any of those pins, since installing the retropie mk_arcade_joystick_rpi driver I can use all them without problems, only not using this because has other problems, like doesn't load my custom pin configuration on boot (I have to do it manual always) and emulationstation doesn't record the mappings I do, keeps saying jpypad detected on every boot.

BaltasarParreira, 2025-03-11:
Ok, final update and solution.
I reboot and stop the gpionext service, them I initiate this time the setup in a normal way not specifyinf my pins, so this:
`gpionext config`

And bonnk !!!,
It worked, all buttons are detected and I manage to save the configuration without any problems,
After doing "gpionext start" the log/journal no long show the information regarding some of the pins I use like before and only shows other ones that probably are really used for either the Rpi system or the I2S sound board I have, but since I have everything working 100% now, not going to bother investigating further that.  ; )